flicr-capstone/issues #18: Research embedded video streaming solutions

mmgoodnow, 2019-08-21:
We may not need to roll our own here. It might be as easy as setting up a config file for an already established video streamer.

mmgoodnow, 2019-08-21:
We should take a look at Motion.

mmgoodnow, 2019-08-28:
@bochola I'm assigning this to you so you can get a little bit more familiar with what we're doing here. 

Basically we think that using a raspberry pi as a surveillance camera or some other sort of live streaming thing is probably already a solved problem, and we want to figure out what is the best solution. 

So just do some googling around, looking for software that does live streaming on a raspberry pi from a camera, and you can report back what you found in a comment here.

mmgoodnow, 2019-08-30:
https://electronicsforu.com/electronics-projects/make-video-streaming-camera-with-raspberry-pi

This one's interesting, uses a vlc command or motion

mmgoodnow, 2019-08-30:
https://raspberrypi.stackexchange.com/questions/23182/how-to-stream-video-from-raspberry-pi-camera-and-watch-it-live

this is the one that Ben found I think, also uses a vlc command

mmgoodnow, 2019-08-30:
this one's old but it seems like this guy knows what he's talking about
https://blog.miguelgrinberg.com/post/stream-video-from-the-raspberry-pi-camera-to-web-browsers-even-on-ios-and-android

mmgoodnow, 2019-08-30:
https://dantheiotman.com/2017/08/23/using-raspivid-for-low-latency-pi-zero-w-video-streaming/

this one says low latency

james-herbstritt, 2019-08-30:
https://hackaday.com/2017/09/12/video-streaming-like-your-raspberry-pi-depended-on-it/

mmgoodnow, 2019-08-30:
So i think we're seeing three main options:
1. `raspivid` piped to something
2. Motion
3. gstreamer

mmgoodnow, 2019-09-16:
@bochola The other thing here that would be great is a way to mock our streaming setup for development and debug purposes. Ideally we can work on the front end and just have it get streams from our computers rather than from the raspberry pi

bochola, 2019-09-16:
Here's all the ones I found that seem mostly applicable. The technical details are still fuzzy for me so I'm not sure which ones will be the most useful or not.

[https://randomnerdtutorials.com/video-streaming-with-raspberry-pi-camera/](https://randomnerdtutorials.com/video-streaming-with-raspberry-pi-camera/)

[https://www.instructables.com/id/How-to-Make-Raspberry-Pi-Webcam-Server-and-Stream-/](https://www.instructables.com/id/How-to-Make-Raspberry-Pi-Webcam-Server-and-Stream-/)

[https://www.hackster.io/sushree-subhasmita-jena/live-cam-with-raspberry-pi-e9f43d](https://www.hackster.io/sushree-subhasmita-jena/live-cam-with-raspberry-pi-e9f43d)

[https://opensource.com/life/15/9/turning-raspberry-pi-portable-streaming-camera](https://opensource.com/life/15/9/turning-raspberry-pi-portable-streaming-camera)

[https://maker.pro/raspberry-pi/projects/raspberry-pi-webcam-robot](https://maker.pro/raspberry-pi/projects/raspberry-pi-webcam-robot)

[https://pi.processing.org/tutorial/camera/](https://pi.processing.org/tutorial/camera/)

mmgoodnow, 2019-10-18:
Here's a good read. https://mux.com/blog/the-low-latency-live-streaming-landscape-in-2019/ Looking into WebRTC now.

mmgoodnow, 2019-10-19:
WebRTC research:

https://www.rs-online.com/designspark/building-a-raspberry-pi-2-webrtc-camera
https://github.com/meetecho/janus-gateway
https://github.com/kclyu/rpi-webrtc-streamer

closed 2020-03-18 by james-herbstritt