flicr-capstone/issues #20: Research motors

james-herbstritt, 2019-08-29:


bochola, 2019-10-01:
James created a excel sheet that's broken up into three sheets focusing on Serial bus, Servos, and DC motors. Over summer 1 we said we would use serial bus for the wheels and servos for the arm, the reason he added DC motors is because they are the cheapest in case we need to change to those to save on cost.

Then he also researched a way to determine the torque value needed which would help us narrow down the motors to get, he found a set of equations (Going to attempt to use them) that provide the required torque, but we'll need the weight from when the thing is empty and full as well as the dimensions of the wheels.

The next steps are for James and I to lock down a sufficient power source that doesn't mess with the weight calculations too much

mmgoodnow, 2019-10-01:
@bochola so what do the rest of us need to do to move this forward?

bochola, 2019-10-01:
@mmgoodnow Right now we need estimates for the total weight of the robot. We know we want it to carry about 5lbs of trash, but how much will the body weigh? It doesn't have to be accurate, though erring on the heavier side will be best I think. The circuits and wire weight will be negligible, its the body, arms, and wheels that matter the most

james-herbstritt, 2020-03-06:
@bochola Can this be closed?

closed 2020-03-18 by james-herbstritt